NVlabs/Deep_Object_Pose #238: Unable to Detect Custom Object/ Produce Belief Maps

BazUCD, 2022-03-29:
Hi I am currently trying to train a custom object (a spanner) for detection with DOPE on a single GPU. After training for 3 Epochs using 20k domain randomized images I was unable to get any detection and I noticed that my loss was far too small from the beginning. 

[image]
the arguments I used for my train.py were as follows:

python train.py --gpuids 0 --batchsize 4 --epochs 3 --data /home/user/Dataset_Synthesizer/Source/NVCapturedData/Spanner1/Spanner1 --namefile Spanner1 --object Spanner1 --outf Spanner

and my object_settings.json is:
[image]
I am unsure of what I am doing wrong and I was also wondering how I can develop some belief maps to see what the training is producing. Any help would be greatly appreciated.

TontonTremblay, 2022-03-29:
I believe the tensorboard of train2.py will have the belief maps. I think the original train might have them as well. Otherwise, I would browse the issues to see how people do it. The code is embedded somewhere on the repo. 

Also without seeing the output I do not think I can help you much. Moreover I do not think I can help much if you do not share training images.

BazUCD, 2022-03-30:
> I believe the tensorboard of train2.py will have the belief maps. I think the original train might have them as well. Otherwise, I would browse the issues to see how people do it. The code is [...]

Hi @TontonTremblay thanks a million for your quick reply. I have attached some new images that I have been using to train in a spanner. What output should I provide as well?
[image]
[image]

BazUCD, 2022-03-30:
[image]
[image]

TontonTremblay, 2022-03-30:
The wrench has a symmetry, so it might be causing confusion. We are working on a PR to help with that. You should check the merge requests. Without more debugging views of what is going on, I am not sure I can help much more.

closed 2022-04-06 by BazUCD