NVlabs/Deep_Object_Pose #243: Inference detecting only objects with 6+ valid point

15ajka, 2022-04-09:
Hello,
thanks for providing great software. I am having problems with inference. Inference script is detecting too little objects, even ones, which look pretty simple to detect. Seems like one of problems may be in PNP solver used in inference. https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/train2/inference/cuboid_pnp_solver.py#L89-L100
In comments it says 4+ valid points should be enough to do the solvePNP. However when I run the code with this if condition, I get an error:

Traceback (most recent call last):
  File "/home/pirhalova/Deep_Object_Pose/scripts/train2/inference.py", line 370, in <module>
    dope_node.image_callback(
  File "/home/pirhalova/Deep_Object_Pose/scripts/train2/inference.py", line 196, in image_callback
    results, beliefs = ObjectDetector.detect_object_in_image(
  File "/home/pirhalova/Deep_Object_Pose/scripts/train2/inference/detector.py", line 423, in detect_object_in_image
    detected_objects = ObjectDetector.find_object_poses(vertex2, aff, pnp_solver, config)
  File "/home/pirhalova/Deep_Object_Pose/scripts/train2/inference/detector.py", line 488, in find_object_poses
    location, quaternion, projected_points = pnp_solver.solve_pnp(points)
  File "/home/pirhalova/Deep_Object_Pose/scripts/train2/inference/cuboid_pnp_solver.py", line 95, in solve_pnp
    ret, rvec, tvec = cv2.solvePnP(
cv2.error: OpenCV(3.4.17) /io/opencv/modules/calib3d/src/calibration.cpp:1099: error: (-2:Unspecified error) in function 'void cvFindExtrinsicCameraParams2(const CvMat*, const CvMat*, const CvMat*, const CvMat*, CvMat*, CvMat*, int)'
> DLT algorithm needs at least 6 points for pose estimation from 3D-2D point correspondences. (expected: 'count >= 6'), where
>     'count' is 5
> must be greater than or equal to
>     '6' is 6

Has something changed since the comment was made ? I am currently using cv2 version: 3.4.17. Should I use different version ? Or is this comment old and I will be only able to detect objects with 6+ point only ? 

E.g. I have this image and inference finds: 3,4,5,6 valid points for one of each instances. That means only one instance is detected in the end. Is this expected behaviour or can I somehow fix this?

[image]

Thanks,

Angelika

DOPE-liyulong, 2022-04-11:
Hello, I'm sorry that I can't answer your question. How do you run the inference script?
Because the error occurred when I ran the inference script (train2/inference. py) :
ERROR: Unable to find model weights: 'weights_dope/resnet_simple/alphabe_soup.pth'

DOPE-liyulong, 2022-04-13:
> 
I solved the problem the day before yesterday.I'm terrible at coding. I made two changes to this script（train2/inference.py）. I don't know if it's right, but it outputs a folder of images and text
1.ERROR:  Unable to find model weights: 'weights_dope/resnet_simple/alphabe_soup.pth'
My method: 'alphabet_soup: "/ home/liyulong zijilianxi_demo_ros/dope_ws/SRC/dope/weights/AlphabetSoup PTH"     replace
https://github.com/NVlabs/Deep_Object_Pose/blob/a4fe3cd4b5a739defdba4818d0a183490237ad0b/scripts/train2/config_inference/config_pose.yaml#L20
2.NameError: name 'ResnetSimple' is not defined
My method：net = DopeNetwork()    replace   https://github.com/NVlabs/Deep_Object_Pose/blob/a4fe3cd4b5a739defdba4818d0a183490237ad0b/scripts/train2/inference/detector.py#L251

It worked

TontonTremblay, 2022-04-13:
I think I had some hooks to load different models as a function of naming, I am glad you found a solution to your problem.

15ajka, 2022-04-14:
Thanks for the tips. As @TontonTremblay suggested, problem was in opencv version. I was using python 3.9 and newest version under 4.3 was 3.4. When I changed python version down to 3.7 and tried different opencv version, I got it to work with 4 points with opencv 4.1.2. Previous version probbably had different implementation of pnp_solve and needed 6 points. Thanks for the tip, now I got to ~80% of objects detected in my synthetic testing dataset compared to previous 15%.

closed 2022-04-14 by 15ajka

TontonTremblay, 2022-04-14:
Thank you for confirming the problem. Good luck. If I get around it I will check to add the right dependency.