NVlabs/Deep_Object_Pose #244: --path_single_obj（nvisii_data_gen/single_video_pybullet.py）.Need help

DOPE-liyulong, 2022-04-16:
Hello, thanks for Martin Günther's contribution, I am studying the recently updated content of symmetric objects. I encountered the following problems:

1. Does ‘--path_single_obj’ specify a training object?

2. I encountered an error when I specified ‘--path_single_obj’:
loading: /home/liyulong/Dataset_Utilities/nvdu/data/ycb/aligned_cm/005_tomato_soup_can/google_16k/textured.obj
Traceback (most recent call last):
  File "single_video_pybullet.py", line 388, in <module>
    adding_mesh_object(f"single_obj_{i_object}",
  File "single_video_pybullet.py", line 371, in adding_mesh_object
    add_cuboid(entity_name, scale=scale, debug=debug)
  File "/home/liyulong/zijilianxi_demo_ros/dopeSymmetry_ws/src/dope/scripts/nvisii_data_gen/utils.py", line 983, in add_cuboid
    child_transform.set_parent(visii.transform.get(f"{name}_symmetry_corrected"))
  File "/usr/local/lib/python3.8/dist-packages/nvisii/__init__.py", line 20246, in set_parent
    return _nvisii.transform_set_parent(self, parent)
RuntimeError: Error: parent is empty
CUDA call (cudaGetDevice(&savedActiveDeviceID)) failed with code 4 (line 145): driver shutting down
terminate called after throwing an instance of 'std::runtime_error'
  what():  fatal cuda error

I don't know if I'm doing something wrong, should I load relative paths?
Should I change the following code to the path of my training object?
https://github.com/NVlabs/Deep_Object_Pose/blob/a4fe3cd4b5a739defdba4818d0a183490237ad0b/scripts/nvisii_data_gen/single_video_pybullet.py#L392-L393

TontonTremblay, 2022-04-20:
This looks good, I think the problem is related to having a parent or not. Can you share with the command you are using to generate data, I will try it and try a fix. This part of the code went through a few updates recently and it might have caused some issues.

DOPE-liyulong, 2022-04-21:
> This looks good, I think the problem is related to having a parent or not. Can you share with the command you are using to generate data, I will try it and try a fix. This part of the code went through a few updates recently and it might have caused some issues.

Thanks for your help, here is the command I used to generate the data:
python single_video_pybullet.py --nb_frames 1 --scale 0.01 --path_single_obj /home/liyulong/Dataset_Utilities/nvdu/data/ycb/aligned_cm/005_tomato_soup_can/google_16k/textured.obj --nb_distractors 0

DOPE-liyulong, 2022-04-21:
Please allow me to ask one more question：Load only one image background during one execution of the script（nvisii_data_gan/single_video_pybullet.py）?    Any help would be appreciated, thank you.

mintar, 2022-04-21:
I must admit that I didn't test my latest pull requests with the `--path_single_obj` option. That option is a bit weird anyway, and I think we should get rid of it. It does things quite differently than the normal `--objs_folder` option. I did my best to make my changes compatible, but I didn't test it, so it's possible I messed something up.

In my opinion, what you should do is create a folder structure like this:

```text
scripts/nvisii_data_gen/models
├── klt
│   └── google_16k
│       ├── model_info.json
│       ├── textured.mtl
│       ├── textured.obj
│       └── texture_map_flat.png
├── materialbox
│   └── google_16k
│       ├── textured.mtl
│       ├── textured.obj
│       └── texture_map_flat.png
├── multimeter
│   └── google_16k
│       ├── textured.mtl
│       ├── textured.obj
│       └── texture_map_flat.png
├── power_drill_with_grip
│   └── google_16k
│       ├── textured.mtl
│       ├── textured.obj
│       └── texture_map_flat.png
├── relay
│   └── google_16k
│       ├── textured.mtl
│       ├── textured.obj
│       └── texture_map_flat.png
└── screwdriver
    └── google_16k
        ├── model_info.json
        ├── textured.mtl
        ├── textured.obj
        └── texture_map_flat.png
```

In other words:
* for each object, there must be a subfolder of `models` with the object name
* there **must** be a subfolder called `google_16k`
* the obj file **must** be called `textured.obj`
* the texture **must** be named `texture_map_flat.png`

Then just point your `--objs_folder` option to that `models` folder. If you use the models folder in the `nvisii_data_gen` directory, you don't need to specify `--objs_folder` at all.

We should also relax these hardcoded paths in the future:
* we should look recursively for the first `obj` file found in each model subfolder and use that; that would get rid of the `google_16k` and `textured.obj` requirements
* we should then look up if there's a `.mtl` file referenced in the `obj` file and read the texture name from the  `.mtl` file; that would get rid of the `texture_map_flat.png` requirement.

But that's future work. For now you'll have to use it as I described above.

TontonTremblay, 2022-04-22:
The mechanism of single object is for simplifying loading any obj file. Nvisii loads an obj file as a set of entities, `import_scene`. Anyway the nature of the scene vs. a single entity is not clean. I will have some time today to push a little update that hacks this part. Stay tuned @DOPE-liyulong. Sorry @mintar I will break your nice 3d model structure :P

TontonTremblay, 2022-04-22:
Can you try the update I made. It is mainly a hack really, but I think it should work for you. If you have multiple objects in your file, please flatten everything to one. 
```
python single_video_pybullet.py --nb_frames 1 --scale 0.01 --path_single_obj models/Ketchup/google_16k/textured.obj --nb_distractors 0 --nb_objects 1 --spp 10 --scale 0.05 
```
This is how I tested it.

DOPE-liyulong, 2022-04-23:
> Can you try the update I made. It is mainly a hack really, but I think it should work for you. If you have multiple objects in your file, please flatten everything to one.
>
> ```
> python [...]

Hello, I tried your command and now I have encountered a new error：
liyulong@liyulong-Lenovo-Legion-Y7000-2020:~/Desktop/dope_ws/src/dope/scripts/nvisii_data_gen$ python single_video_pybullet.py --nb_frames 1 --scale 0.01 --path_single_obj models/Ketchup/google_16k/textured.obj --nb_distractors 0 --nb_objects 1 --spp 10 --scale 0.05 
pybullet build time: Dec  1 2021 18:33:43
folder output/output_example// exists
Warning, create_perspective_from_fov is deprecated and will be removed in a subsequent release. Please switch to create_from_fov.
loading: models/Ketchup/google_16k/textured.obj
Traceback (most recent call last):
  File "single_video_pybullet.py", line 405, in <module>
    adding_mesh_object(f"single_obj_{i_object}",
  File "single_video_pybullet.py", line 388, in adding_mesh_object
    add_cuboid(entity_name, scale=scale, debug=debug,toy_transform = toy_transform.get_name())
TypeError: add_cuboid() got an unexpected keyword argument 'toy_transform'
CUDA call (cudaFree((void*)d_pointer)) failed with code 4 (line 94): driver shutting down
terminate called after throwing an instance of 'std::runtime_error'
  what():  fatal cuda error
Aborted (core dumped)

I wonder if I did something wrong? I have downloaded the latest DOPE.

TontonTremblay, 2022-04-23:
just pushed the fixed, sorry it was a hack that did not work.