NVlabs/Deep_Object_Pose #245: Publish "NaN" values when object not detected

erikkockar, 2022-04-17:
Hi guys, 

I have a question regarding publishing values. 
If I do not detect any object in current image I would like to always publish some specific values. 

So for example my object of interest is not detected in the current image, DOPE will be still publishing some hardcoded data for example zeros. Where should I take look in code i am a bit lost. This is maybe more ROS question maybe there is some easy way ? 

Thanks

mintar, 2022-04-18:
DOPE already does that! For every input image that DOPE processes, one detected_objects message is published. If no objects are detected, the detected_objects array is empty.

TontonTremblay, 2022-04-18:
https://github.com/NVlabs/Deep_Object_Pose/blob/master/nodes/dope#L260 specifically the logic is here, you can see the messages getting created per object.

closed 2022-05-04 by erikkockar