NVlabs/Deep_Object_Pose #246: Unable to load object from YCB dataset

rahulram70, 2022-04-18:
Hello.

I'm trying to generate custom dataset using objects from the YCB dataset. To do so, I downloaded the Gelatin Box google 16k files from http://ycb-benchmarks.s3-website-us-east-1.amazonaws.com/ and put them in the ~/catkin_ws/src/dope/scripts/nvisii_data_gen/models directory. I then ran the following command "python single_video_pybullet.py --nb_frames 1 --scale 0.01". 

The terminal states the Gelatin Box and Ketchup was loaded. The ketchup bottles appear in the image, but the Gelatin Box is now where to be found. There is also no entries of the Gelatin Box in the corresponding json. I then tried loading the gelatin textured.obj via the --path_single_obj parameter, but all I got was RuntimeError: Error: parent is empty.

If it helps, I'm running this on an Ubuntu 20.04 machine. Any help would be appreciated, thank you.

rahulram70, 2022-04-18:
Yes I did, and it didn't work. Even if my object did not appear in the image due to the scale, there should still be an entry in the json file, correct?

TontonTremblay, 2022-04-18:
Did you try `--interactive` ?

rahulram70, 2022-04-21:
I tried, and I couldn't find the tennis ball. I reverted to a previous commit and was able to successfully load the YCB objects.

closed 2022-04-21 by rahulram70

mintar, 2022-04-21:
So what was the previous commit where it worked?

Regarding the `RuntimeError: Error: parent is empty` with `--path_single_obj`: That's an unrelated problem and discussed in #244.

rahulram70, 2022-04-21:
I switched to commit a25e028 since I didn't need object symmetry. I did not try the numerous commits after a25e028.