NVlabs/Deep_Object_Pose #253: Too slow on YCB dataset

kooshyarkosari, 2022-05-11:
Hi

I use the following code in order to see the result of the Dope on the YCB video Dataset ( taking the prerecorded video as input ), everything goes well But I have a problem with the processing time ( 1.7011144739999509  seconds) per frame, which is too slow !!! 

real    26m19.766s
user    25m36.176s
sys     6m49.752s

 So what can I do to solve this problem?

 **Please note that I tested on Ubuntu 16.04.5  with an NVIDIA Tesla K80 with Python 3.8.** 

    lineWidthForDrawing = 2

    # draw front
    DrawLine(points[0], points[1], color, 3)
    DrawLine(points[1], points[2], color, 3)
    DrawLine(points[3], points[2], color, 3)
    DrawLine(points[3], points[0], color, 3)

    # draw back
    DrawLine(points[4], points[5], color, 1)
    DrawLine(points[6], points[5], color, 1)
    DrawLine(points[6], points[7], color, 1)
    DrawLine(points[4], points[7], color, 1)

    # draw sides
    DrawLine(points[0], points[4], color, 2)
    DrawLine(points[7], points[3], color, 2)
    DrawLine(points[5], points[1], color, 2)
    DrawLine(points[2], points[6], color, 2)

    # draw dots
    DrawDot(points[0], pointColor=color, pointRadius=4)
    DrawDot(points[1], pointColor=color, pointRadius=4)

    # draw x on the top
    DrawLine(points[0], points[5], color, 1)
    DrawLine(points[1], points[4], color, 1)

    models = {}
    pnp_solvers = {}
    pub_dimension = {}
    draw_colors = {}

    # Initialize parameters
    matrix_camera = np.zeros((3,3))
    matrix_camera[0,0] = params["camera_settings"]['fx']
    matrix_camera[1,1] = params["camera_settings"]['fy']
    matrix_camera[0,2] = params["camera_settings"]['cx']
    matrix_camera[1,2] = params["camera_settings"]['cy']
    matrix_camera[2,2] = 1
    dist_coeffs = np.zeros((4,1))

    if "dist_coeffs" in params["camera_settings"]:
        dist_coeffs = np.array(params["camera_settings"]['dist_coeffs'])
    config_detect = lambda: None
    config_detect.mask_edges = 1
    config_detect.mask_faces = 1
    config_detect.vertex = 1
    config_detect.threshold = 0.5
    config_detect.softmax = 1000
    config_detect.thresh_angle = params['thresh_angle']
    config_detect.thresh_map = params['thresh_map']
    config_detect.sigma = params['sigma']
    config_detect.thresh_points = params["thresh_points"]

    # For each object to detect, load network model, create PNP solver, and start ROS publishers
    for model in params['weights']:
        models[model] = \
            ModelData(
                model,
                "weights/" + params['weights'][model]
            )
        models[model].load_net_model()

        draw_colors[model] = tuple(params["draw_colors"][model])

        pnp_solvers[model] = \
            CuboidPNPSolver(
                model,
                matrix_camera,
                Cuboid3d(params['dimensions'][model]),
                dist_coeffs=dist_coeffs
            )

    vidcap = cv2.VideoCapture('YCB.mp4')
    fourcc = cv2.VideoWriter_fourcc(*'XVID')
    out = cv2.VideoWriter('output.avi', fourcc, 20.0, (640,  480))
   
    length = int(vidcap.get(cv2.CAP_PROP_FRAME_COUNT))
    print( length )
    print('The Total Frame:', length)

    while (vidcap.isOpened()):

mintar, 2022-05-11:
The more object classes you load, the slower DOPE becomes. So the easiest way to speed it up is to reduce the number of object classes. Additionally, you can reduce the number of stages or the resolution of the input image, but both will degrade accuracy.

TontonTremblay, 2022-05-12:
You could also batch the videos, this is how we did it for the paper, but that code did not make it to the repo. The idea is that you would load 32 images and then run the dope network on them, then you would run pnp as linearly. This last step should be pretty fast. Computing ADD can also be costly specially if you use a large point cloud.

closed 2022-05-18 by kooshyarkosari