NVlabs/Deep_Object_Pose #254: How to create models_info file ?

Charan0502, 2022-05-15:


mintar, 2022-05-16:
See the README: https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/readme.md#handling-objects-with-symmetries

Charan0502, 2022-05-18:
Hello sir,
Can you kindly help me in creating models_info file for this model

https://drive.google.com/file/d/1-vdc7o-pCLXg0vD-Au-LCLehnlgIrF7a/view?usp=sharing

mintar, 2022-05-31:
First off, the object center should be shifted to coincide with the center of the bounding box. That way, you can leave the `offset` at `[0, 0, 0]`.

The top and bottom look different, so you don't need a 180° discrete symmetry. You probably want to treat your object as a cylinder with a rotational axis of y, so the model_info.json should be:

```json
{
  "symmetries_continuous": [{"axis": [0, 0, 1], "offset": [0, 0, 0]}],
  "align_axes": [{"object": [1, 0, 0], "camera": [0, 0, 1]}]
}
```