NVlabs/Deep_Object_Pose #258: Can't find pyrr

rahulram70, 2022-06-02:
Hello. 

When running  roslaunch dope dope.launch, I get cannot find module pyrr. This is quite odd since I have it installed, and I'm able to import it in python.

I have provided a screenshot of the issue if that helps. Thank you in advance.

[image]

rahulram70, 2022-06-02:
Turns out I had an issue with my environment.

closed 2022-06-02 by rahulram70

Jeong-bhin, 2023-10-31:
hi, 
I also faced the same issue. I would like to know what the problem was. Thank you in advance!

TontonTremblay, 2023-10-31:
I think this might be a problem with your ros python not having pyrr. I had this issues as well. Do you know the path of your ros python? If so use its pip to install pyrr.

Jeong-bhin, 2023-11-01:
thank you for your comment!

It seems that the issue was related to not installing it in the ROS Python because I'm using a Conda environment. I resolved the problem with the following code.

```shell
sudo -H pip install pyrr
``` 
However, I encountered another problem. I am using PyTorch version 1.6.0 in Conda environment, but it is not recognizing it. I would appreciate it if you could provide a solution to this issue. I'm sorry for taking your time.
[image]

Jeong-bhin, 2023-11-01:
I additionally checked the Torch version in Conda environment.
[image]

TontonTremblay, 2023-11-01:
I am not 100% sure since I am not a pro with ros, but when I had this problem, the python path catkin was build with was wrong, something you can do is delete or rename the build folder, and then redo catkin_make, but with your conda python env enabled. This will tell ros what your python path is. https://github.com/ms-iot/ROSOnWindows/issues/67 this has some of the logic what I am trying to explain.

TontonTremblay, 2023-11-01:
when you run the script you can always look at
```
import sys

print(sys.executable)
```
that will tell you which python is being used. But I think doing a clean catkin with your conda env enabled should fix this.

Jeong-bhin, 2023-11-02:
I tried as per your suggestion. First, I used ```catkin clean```  to clean everything up, and then I ran ```catkin make``` and it worked well. Thank you for your help in resolving the issue!!

TontonTremblay, 2023-11-02:
python in ros is a nightmare.

guohaot, 2023-11-23:
> I tried as per your suggestion. First, I used `catkin clean` to clean everything up, and then I ran `catkin make` and it worked well. Thank you for your help in resolving the issue!!
Firstly I am sorry for taking your times.  I encountered the same problem，one is the pyrr，Then I resolved the problem by yours code.  
`sudo -H pip install pyrr
`
But the second problem about No module named 'torch'，I run `catkin clean` ，but it seems not resolute it，it turn out like this：

[image]
it need the arg “-w”,but I don't understand what this parameter needs to be filled in.

[image]
My workspace is **catkin_ws**, I try to follow  “-w **./** **build** **devel** ” ,It can't solve the problem

TontonTremblay, 2023-11-23:
I am sorry my understanding of ros is quite limited, I would try chagpt or bard at this moment if I were you.