NVlabs/Deep_Object_Pose #259: Unable to generate images on NVIDIA GK110GL [Quadro K6000]

utsavrai, 2022-06-08:
Hi, 
When running the single_video_pybullet.py on the Dell Precision Tower 7810 equipped with NVIDIA Corporation GK110GL [Quadro K6000], it throws following error - 
```
python3 single_video_pybullet.py --nb_frames 10
pybullet build time: May 20 2022 19:44:17
folder output/output_example// exists
Optix call (optixDeviceContextCreate(cudaContext, 0, &optixContext)) failed with code 7051 (line 207)
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
Aborted (core dumped)
```
Please do let me know if I could run the script on my hardware.

Thanks

mintar, 2022-06-08:
Unfortunately, that sort of error usually means that your GPU isn't supported by Optix. :(

utsavrai, 2022-06-12:
Okay, thank you