NVlabs/Deep_Object_Pose #264: Use training script train2 for NDDS data

WilkoLu, 2022-08-15:
Hello

I need NDDS Data to work on the "new" training script with tensorboard visualisation. It seems not possible yet. Do you consider to improve in this way? I want to use NDDS data for training and NVISII data for testing. NDDS is pretty performant in creating big datasets for training and NVISII delivers good looking testing datasets.

TontonTremblay, 2022-08-24:
Did you try? @mintar I think made everything backward compatible!

mintar, 2022-08-25:
I had to check myself to see what I did. This is what I did:

* The old script (`train.py`) can now process both NDSS and NVISII data.
* The new script (`train2/train.py`) can only process NVISII data.

The second point is @WilkoLu 's problem.

However, I think the differences are small, so it should be easy to adapt `train2` to the NDSS format. Here is what's currently failing:

* "visibility" must be > 0, but it's always 0 in my NDSS data: https://github.com/NVlabs/Deep_Object_Pose/blob/538d12bf1aef17d3f094eaefe4f82d46d69b0575/scripts/train2/utils_dope.py#L227
* "projected_cuboid" must have 9 points (8 cuboid corners + centroid); the NDSS data has 8 points, and the field "projected_cuboid_centroid" is stored separately: https://github.com/NVlabs/Deep_Object_Pose/blob/538d12bf1aef17d3f094eaefe4f82d46d69b0575/scripts/train2/utils_dope.py#L228

Here we could check whether we have 8 or 9 points in the json file, and if it's 8, read "projected_cuboid_centroid" and append it, like I did in the old `train.py` script:

https://github.com/NVlabs/Deep_Object_Pose/blob/538d12bf1aef17d3f094eaefe4f82d46d69b0575/scripts/train.py#L294-L300

I think in the long run we should unify `train.py` and `train2`, there's no point in keeping both around.