NVlabs/Deep_Object_Pose #267: Install ROS dependencies for Windows

HripsimeS, 2022-09-15:
@mintar @TontonTremblay @sbirchfield @nv-jeff @blaine141 @520xyxyzq @hemalshahNV @Ademord @balakumar-s @v4hn @JRauer

Dear NVlabs team,

I'm trying to use DOPE with ROS inference to do 3D object detection by using IntelRealSense camera. I followed all the instuctions 
Install ROS with this instruction http://wiki.ros.org/Installation/Windows
Create a catkin workspace, Download the DOPE code, Install python dependencies. But on the stage of Install ROS dependencies I received an error connected to OS (I use Windows), please find the screenshot of error below. 

Is there any way to install ROS dependencies and make your method work in Windows OS or it works only with Ubuntu ??

[image]

sbirchfield, 2022-09-15:
I don’t think we’ve run it on Windows yet.  Should be possible, but maybe require some extra steps.

Adding @Jeffrey ***@***.***> in case he has thoughts.

--Stan

[dope]<https://user-images.githubusercontent.com/42246765/190342886-f5a0421a-df8e-4ed2-9af9-cbdc09ba9b63.png>

HripsimeS, 2022-09-15:
@sbirchfield thanks for your quick reply! I installed Ubuntu with Vmware virtual machine and followed the instruction for ROS Noetic installation http://wiki.ros.org/noetic/Installation/Ubuntu

But even with Ubuntu OS I still received errors and was unable to finish the installalation, please find it below.
Any ideas what is going wrong with ROS Noetic installation? 

[image]

HripsimeS, 2022-09-16:
@mintar @TontonTremblay @sbirchfield @nv-jeff @blaine141 @520xyxyzq @hemalshahNV @Ademord  @balakumar-s 

Hello. I have installed Ubuntu 20.04 LTS on WSL2 on Windows 10 and thanks to that I could install everything until the stage of installation of ROS dependencies when I got the error that dope cannot locate rosdep definition for [camera_info_manager_py]. 
Do you know what is the issue and hox to fix it? Thanks in advance for your help!

[image]

mintar, 2022-09-16:
You have to edit `package.xml` and remove the line `<depend>camera_info_manager_py</depend>`. Also see #167. Sorry for the inconvenience.