NVlabs/Deep_Object_Pose #273: Not able to visualize 3D cuboid in rviz

monajalal, 2022-11-30:
I tried both of these and none works for publishing the 3D cuboid when I select the "Image" in "Add --> Display Name" however the actual image and the belief maps are published. Can you please guide me how to see the 3D cuboid around the meat can? In config_pose.xml I have commented soup can and set to meat.

~/catkin_ws/src/camera_info_manager_py# rosrun tf2_ros static_transform_publisher 0 0 0 0.7071 0 0 -0.7071 world
[ INFO] [1669839875.757842505]: Spinning until killed publishing -0.7071 to world
~/catkin_ws/src/camera_info_manager_pyrosrun tf static_transform_publisher 0 0 0 0 0 0 1 map dope_webcam 5

[image]

[image]

Here's the config_pose.xml change:

```
    # "cracker":"package://dope/weights/cracker_60.pth",
    # "gelatin":"package://dope/weights/gelatin_60.pth",
     "meat":"package://dope/weights/meat_20.pth",
    # "mustard":"package://dope/weights/mustard_60.pth",
    #"soup":"package://dope/weights/soup_60.pth",
    #"sugar":"package://dope/weights/sugar_60.pth",
    # "bleach":"package://dope/weights/bleach_28_dr.pth"

```

TontonTremblay, 2022-12-01:
Can you check by topics? I am not sure what is going on here. I am not sure the detection is working correctly. I see some heatmap, but no detection on the second image, e.g., bounding box.

monajalal, 2022-12-01:
I have tested almost everything in topic and nothing shows the pose. Could you please show which one do you select? I haven't changed your code and I though it would work for the items and it seems people are getting it to work without change of code.

When I first open rviz I get a warning for "global status"
[image]

Then I use the command at the end of your README.md for fixed frame but it throws an error:

`~/catkin_ws# rosrun tf2_ros static_transform_publisher 0 0 0 0.7071 0 0 -0.7071 world`
[image]

Then I changed it to the following command and the error is gone however still no pose shown.
`~/catkin_ws# rosrun tf static_transform_publisher 0 0 0 0 0 0 1 map dope_webcam 5`
[image]

[image]

Here are what pose_meat and PoseArray show:
[image]

For the image that nothing shows ('No Image') here's the side panel. This is the image selected by 'Display Name'. The image that has a SPAM is from topic --> webcam--> image
[image]

Manu752, 2025-05-16:
hi I am having the same issue . Were you able to fix this ?

Manu752, 2025-05-18:
Hi @TontonTremblay  , Thanks you for  replying . I have attached my rviz image .. I am using the weights for the ketchup ie:"Ketchup.pth" . I also  get the following result in my terminal window -  " Model loaded in 1.6927964687347412 seconds.
Running DOPE...  (Listening to camera topic: '/dope/webcam/image_raw')
Ctrl-C to stop
Incomplete cuboid detection.
  result from detection: [None, None, None, None, None, None, None, None, (267.02903125475336, 276.28448924721044)]
Skipping."

[image]

[image]