NVlabs/Deep_Object_Pose #274: Question on the belief map dimension

Juphex, 2022-12-05:
I am trying to understand the model but I am note quite sure about the belief maps. I know why the dimension of the vector fields are 1x1x16. 
But why are the belief maps given as 1x1x9?
When there is a belief map for each vertex and the centroid (8+1) why is the dimension for each map 1x1?

Greetings

TontonTremblay, 2022-12-05:
they should be a factor of the initial image resolution, e.g., 400x400x4 -> 50x50x9 or 50x50x16. I am not sure how you are getting the 1s.

TontonTremblay, 2022-12-05:
> Thanks!
> I was just wondering because in the paper it was called "1 × 1 × 16
> layer". But now the sentence after that make sense for me. "All stages are
> of size w/8 and h/8".

Sorry about the confusion, I hope we were able to clear this out.