NVlabs/Deep_Object_Pose #276: training without dealing with ROS

monajalal, 2022-12-08:
I am interested in standalone and independent from ROS training. I noticed you have two scripts for that:

scripts/train2/train.py
and 
scripts/train.py

*I was also able to train the shiny meat dataset with train2/train.py with no problem.

I used the second one and got error for the code not being able to find the `make_grid` by having the original line `from dope.utils import make_grid` so I changed it to `from train2.utils_dope import make_grid` and didn't receive any errors. Is this method acceptable or do you have a better suggestion for fixing the error?
```
import cv2
import sys
sys.path.append("/home/mona/research/dope/scripts/train2")

import colorsys

#from dope.utils import make_grid
from train2.utils_dope import make_grid
```

and here's the result:
```
~/research/dope/scripts$ python train.py  --data MYDATA --object MYOBJECT --outf MYOUTDIR--batchsize 16
start: 20:34:53.369886
load data
training data: 1563 batches
load models
Training network pretrained on imagenet.
Train Epoch: 1 [0/25002 (0%)]   Loss: 0.042984262108803
Train Epoch: 1 [1600/25002 (6%)]        Loss: 0.005995919927955
Train Epoch: 1 [3200/25002 (13%)]       Loss: 0.006112941075116
Train Epoch: 1 [4800/25002 (19%)]       Loss: 0.006253230851144
```

Other side questions:
1. can `scripts/train2/train.py` and  `scripts/train.py` be used for any dataset that is made with either of NDDS or NViSii?
2. Is there a preference to use one over the other one?

mintar, 2022-12-09:
> Is this method acceptable or do you have a better suggestion for fixing the error?

I think it's okay, although the real path of the file that `scripts/train.py` is supposed to be using is `src/dope/utils.py`.

> can `scripts/train2/train.py` and `scripts/train.py` be used for any dataset that is made with either of NDDS or NViSii?

`scripts/train.py` can handle NDDS and NViSii format. `scripts/train2/train.py` can only handle NViSii format.

> Is there a preference to use one over the other one?

There are some minor differences, but I haven't been able to figure out yet if they make a significant difference one way or the other. I've used both scripts recently with decent results. In the long term, we should clean up the code base to only have one training script.

monajalal, 2022-12-09:
Thank you so much. This makes sense why I was not able to train the train2 for a NDDS3 dataset. So far, this answered my question so I close the issue for now.

closed 2022-12-09 by monajalal