NVlabs/Deep_Object_Pose #278: training on FAT single meat NDDS format doesn't produce good heatmap hence no object pose

monajalal, 2022-12-20:
My intention is to show a demo of full training from scratch and then getting inference for single meat object. I trained the single meat from the scratch using dope/script/train.py (with the modification to account for distributed data parallel) (code below).

I used the single meat (6000 images) from FAT dataset from your DOPE paper.
I used one of the training images (you would expect overfitting and good results since it already has seen it) however the heatmap is very bad.

I used both the original camera_info.yaml and config_pose in `~/research/dope/scripts/train2/config_inference` as well as the one I modified.

Could you please let me know how I can use your dataset or pth for getting an inference on meat object that would work with NDDS format or please provide the visualization code for HOPE dataset that has the NViSii format for meat? 

Here's the command I used for DDP training:
```
azureuser@mona-pose-4gpu:~/research/dope/scripts$ time python -m torch.distributed.launch --nproc_per_node=4 train.py --data ~/010_potted_meat_can_16k --object meat --outf ~/single_NDDS_meat/ --gpuids 0 1 2 3 --batchsize 16 --manualseed 42 --workers 16
```

```
azureuser@mona-pose-4gpu:~/research/dope/scripts/train2$ python inference.py --data test_meat/
Performing interactive authentication. Please follow the instructions on the terminal.
To sign in, use a web browser to open the page https://microsoft.com/devicelogin and enter the code MYAZURECODE to authenticate.
Interactive authentication successfully completed.
output is located in out_experiment
{'meat': '/home/azureuser/single_NDDS_meat/net_epoch_60.pth'}
*****
model is:  meat
config['architectures'] :  {'pudding': 'dope', 'alphabet_soup': 'resnet_simple', 'meat': 'dope'}
Loading DOPE model '/home/azureuser/single_NDDS_meat/net_epoch_60.pth'...
    Model loaded in 37.65599083900452 seconds.
loaded
Ctrl-C to stop
opt data:  test_meat/
i_image is:  0
len(imgs):  2
frame 000000.left.jpg
frame type is <class 'numpy.ndarray'> and frame length 540
vertex2 size[0]:  9
vertex2 size:  torch.Size([9, 50, 88])
peaks len:  12
peaks len:  8
peaks len:  9
peaks len:  3
peaks len:  0
peaks len:  5
peaks len:  0
peaks len:  15
peaks len:  13
len(all_peaks[-1]):  13
nb_object:  0
all_peaks[-1][nb_object][2]:  0.0006037783
nb_object:  1
all_peaks[-1][nb_object][2]:  0.0012903158
nb_object:  2
all_peaks[-1][nb_object][2]:  0.0011796076
nb_object:  3
all_peaks[-1][nb_object][2]:  0.00018327497
nb_object:  4
all_peaks[-1][nb_object][2]:  0.0006824564
nb_object:  5
all_peaks[-1][nb_object][2]:  0.0006898381
nb_object:  6
all_peaks[-1][nb_object][2]:  0.0007104799
nb_object:  7
all_peaks[-1][nb_object][2]:  0.00019483268
nb_object:  8
all_peaks[-1][nb_object][2]:  0.0023401994
nb_object:  9
all_peaks[-1][nb_object][2]:  0.0017732698
nb_object:  10
all_peaks[-1][nb_object][2]:  0.0010225624
nb_object:  11
all_peaks[-1][nb_object][2]:  0.0029813554
nb_object:  12
[... 1565 lines cut ...]

    if not opt.nbupdates is None and nb_update_network > int(opt.nbupdates):
        break

cleanup()

print ("end:" , datetime.datetime.now().time())
```

left belief map:

[image]

right belief map:
[image]

original left image:

[image]

original right image:

[image]

Thanks a lot for any input.

TontonTremblay, 2023-01-04:
The SPAM is quite hard to train. I have had training sessions that did not produce anything. Are the weights we provided not working for you?

monajalal, 2023-01-05:
Thanks for your response. My goal is to do an end-to-end training for meat object, making sure everything works, and then move on to my single object of interest. 

I chose this data since I wanted to be able to visualize the inference results.

When I chose the Shiny Meat pth trained model, I was not able to visualize it using nvdu pose visualizer since that data is created with NViSii not NDDS.

Would you please be able to share or open-source your pose visualizer code for the inference results on Shiny Meat or any other NViSii format dataset?

monajalal, 2023-05-09:
just going back to this, is there any way to reproduce some of the results of your original DOPE paper? for example, is trying the Cheez-It cracker box for object training using scripts/train.py would work? @TontonTremblay

TontonTremblay, 2023-05-09:
Using the weights you should be able to reproduce the numbers, but I am sorry I do not have the training data anymore.

monajalal, 2023-05-09:
Out of curiosity, didn't you use the entire DOPE dataset for this? If I am training using the entire FAT dataset for cracker box, do you expect to receive any meaningful result?

TontonTremblay, 2023-05-10:
yeah it should work fine. The paper shares some results with using FAT. I don't remember if it is all FAT or FAT where the cracker is in the scene.