NVlabs/Deep_Object_Pose #279: Need help for nvisii_data_gen

gary13139948, 2022-12-26:


gary13139948, 2022-12-26:
Hi,
I'm trying to generate synthetic data using single_video_pybullet.py.
However, I encountered a pybullet.error as following.
The error code said: invalid mesh filename './', but there isn't any './' in my .obj file.
What problem might I encounter?

Thanks
[image]

TontonTremblay, 2023-01-04:
Could you share your 3d model with me? I think there is an issue with passing the vertices to create a shape for your object. I should have added a simple use cuboid volume if it crashes. I will test this out later today.

gary13139948, 2023-01-05:
[google_16k.zip](https://github.com/NVlabs/Deep_Object_Pose/files/10351924/google_16k.zip)

> Could you share your 3d model with me? I think there is an issue with passing the vertices to create a shape for your object. I should have added a simple use cuboid volume if it crashes. I will test this out later today.

Sure, please check the attachment.

TontonTremblay, 2023-01-06:
nvisii wont load vertex color correctly I think 
[image]
I think you need to update the code, I had put a guard to avoid that error. 

[image]

I was able to render your model, but you will need to use a texture instead of index colors. You could try that, https://steamcommunity.com/sharedfiles/filedetails/?id=1265761945