NVlabs/Deep_Object_Pose #286: How can we use DOPE without camera

Yukizyh, 2023-03-06:
Hi, I'm using wsl2 ubuntu 20.04 and I've tried using _usbipd_ to connect to a usb camera but it doesn't work, so I'm wondering if it's possible to use DOPE without a camera node? How do I do this, thanks!

TontonTremblay, 2023-03-06:
You could use DOPE with images. Did you try the folder no_ros_inference?

Yukizyh, 2023-03-06:
Hi, But I didn't find the no_ros_inference folder.

TontonTremblay, 2023-03-07:
https://github.com/NVlabs/Deep_Object_Pose/tree/master/scripts/train2

Sorry it was as the internal name. Check this part.

Yukizyh, 2023-03-07:
Thanks for replying. I tried the inference.py but there is some error:

[image]

I have tried changing the path to weight and using other objects like 'soup', but they all failed for the same reason : ERROR:  Unable to find model weights: 'weights_dope/resnet_simple/alphabe_soup.pth'. I also tried to use the whole path of the weight, it doesn't work. And there is a warning here that the camera cannot be opened. I would like to know what are the changes that need to be made.

Yukizyh, 2023-03-07:
I created the path that matches the weight path in config file. So, it works now. Thanks a lot.
But there is an error that shows up when I try to generate the dataset:
[image]
And my cuda is 12.1. Does this problem cause by the wrong version cuda?

TontonTremblay, 2023-03-07:
I will refer you on how to install nvisii, you need an NVIDIA GPU with up to date drivers.

Yukizyh, 2023-03-13:
> u

Thanks a lot! Where can I find the reference?