NVlabs/Deep_Object_Pose #287: The output of the pre-trained CNN

Linhhongduc, 2023-03-13:
Hello, I have successfully used the DOPE model and am now in the process of understanding its code and how it works. I am wondering what the output of the pre-trained CNN for the ObjectDetector class is? Is it tthe 2D location of keypoints on the object, its vertices, or something else?

TontonTremblay, 2023-03-13:
yeah exactly. It is the 9 keypoints that define the object in 3d, but the output is the projected points in the image plane.

Linhhongduc, 2023-03-14:
The output of CNN is the input of ObjectctDetector class ( 3D keypoints), and the output of OD class is projected points, right? Because I find def projected point in this class.

TontonTremblay, 2023-03-14:
projected keypoint is the output of the network, 3d keypoints are generated from the object sizes, the output of dope is the final pose using pnp.

Linhhongduc, 2023-03-14:
https://user-images.githubusercontent.com/124419166/225162512-be7aa12d-a6f1-4967-8b9f-18b89f5865bc.mp4

Thank you very much.  I finished streaming video from MFC on windows over wsl using tcp/ip and measuring depth.  This is a great topic for me to learn and develop😁

TontonTremblay, 2023-03-14:
Hey these results look great :D

TontonTremblay, 2023-03-14:
Thank you for sharing!

Linhhongduc, 2023-03-14:
Thank you.  I will try to help anyone who has problems using WSL virtual machine like me :D

closed 2023-06-24 by Linhhongduc