NVlabs/Deep_Object_Pose #288: KeyError: 'fixed_model_transform'

ArghyaChatterjee, 2023-03-16:
Hello, I was trying to get some synthetic data generated using nvisii. I wanted to see the bounding box using nvdu. Here is the error that I am getting:
```
arghya@arghya-All-Series:~/ros1_catkin_make_ws/src/dope/scripts/nvisii_data_gen/output/output_example$ nvdu_viz
args: Namespace(auto_change=False, auto_export=False, camera_settings_path=None, data_annot_dir='', dataset_dir='.', export_dir='', fps=10, ignore_fixed_transform=False, model_dir='/home/arghya/Dataset_Utilities/nvdu/data/ycb/original', name_filters=['*.png'], object_settings_path=None, size=[0, 0])
name_filters: ['*.png']
Traceback (most recent call last):
  File "/home/arghya/.local/bin/nvdu_viz", line 11, in <module>
    load_entry_point('nvdu', 'console_scripts', 'nvdu_viz')()
  File "/home/arghya/Dataset_Utilities/nvdu/tools/test_nvdu_visualizer.py", line 91, in main
    dataset_settings = DatasetSettings.parse_from_file(object_settings_path, model_dir_path)
  File "/home/arghya/Dataset_Utilities/nvdu/core/nvdu_data.py", line 296, in parse_from_file
    parsed_settings = cls.parse_from_json_data(json_data, mesh_dir_path)
  File "/home/arghya/Dataset_Utilities/nvdu/core/nvdu_data.py", line 275, in parse_from_json_data
    obj_initial_matrix = Matrix44(check_obj['fixed_model_transform'])
KeyError: 'fixed_model_transform'
```
Can you tell me what's the problem ? Looks like the problem is coming from [here].(https://github.com/NVIDIA/Dataset_Utilities/blob/master/nvdu/core/nvdu_data.py) 

N.B: I can't see any issue creating a template or option in the actual repo of nvdu[https://github.com/NVIDIA/Dataset_Utilities].

ArghyaChatterjee, 2023-04-04:
Non Photorealistic data generation using NViSII is causing the error. When used photo realistic data generation pipeline from UE4, I can visualize the annotations correctly but data which were generated by NViSII can't be seen using nvdu. 

Is there any particular reason for this ? Also I see different formats in 2 different dataset. One is having slightly different oriented  annotation format than the other one. 

How do you combine these unmatched dataset during training ? 
And also, have you tried to crosscheck the annotations generated by NViSII using NVDU ? How do you do that ?
Are you combining photorealistic dataset with non photorealistic dataset at all during training or not ?

TontonTremblay, 2023-04-05:
nvisii data generation is not met for compatibility with nvdu. I think there is an other issue describing how to generate the missing data. Sorry.