NVlabs/Deep_Object_Pose #292: Some questions regarding automatic labeling using NViSII

ArghyaChatterjee, 2023-04-02:
Hi,

I have 3 questions. They are:
1. During data generation, we also generate depth.exr and seg.exr files. Do we need them for training purpose or they are just there for visualization purpose? If we don't need them, then how to stop generating them during automatic label generation process?

2. Inside the json files created automatically using NViSII, they contain the distractors pose as well as the pose of the object to be detected. Do we need the distractors pose inside that file for training purpose or they are just there. If we don't need them, how to disable generating them from the beginning ?

Thanks in advance.

TontonTremblay, 2023-04-02:
segmentation: https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/single_video_pybullet.py#L584-L592
depth: https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/single_video_pybullet.py#L613-L621
Wrap this https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/single_video_pybullet.py#L394-L396 with a if, pass to the method something like `to_be_included` and set it to false for these objects: https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/single_video_pybullet.py#L408

ArghyaChatterjee, 2023-04-03:
@TontonTremblay , thanks. BTW, do I need the poses of these distractors during training ?

TontonTremblay, 2023-04-03:
you do not, so in the above you can remove them from the list. If the above
text is unclear, please let me know.