NVlabs/Deep_Object_Pose #293: Getting access to the annotated image and json data on which the original models were trained on

ArghyaChatterjee, 2023-04-06:
Hi,

I am trying to generate or replicate the actual result showed in the paper. So, for that, I think you have generated dataset with different background both photorealistic and non-photorealistic. Can I have access to the training data that you used ?

Thanks,
Arghya

TontonTremblay, 2023-04-06:
I am so sorry, I do not have access to the data anymore. We were not really
precise with our data back then.

ArghyaChatterjee, 2023-04-06:
Ah, ok. Thanks for letting me know. Also, how did you split the photorealistic (from UE4) and non-photorealistic (NViSII generated) dataset during training ? Say in 100k dataset for a single object, how many were photorealistic and how many were non-photorealistic ?

TontonTremblay, 2023-04-07:
Part of the dataset is available online it is the FAT dataset.

From what I remember it was 60k from FAT (selected randomly) and 60k from
domain randomization all rendered with UE4.

For the HOPE object I did 60k from NViSII script from this repo and not FAT
dataset for that one.

joansaurina, 2024-07-01:
Hey @TontonTremblay
And for the YCB objects? 
For mustard I'm using 75.000 blenderproc images with mean results. Should a combination with FAT give better results?
Thanks,
Joan

TontonTremblay, 2024-07-03:
For the YCB it was 60k from FAT (selected randomly) and 60k from domain randomization all rendered with UE4 (NDDS).

The FAT was also built for stereo cameras (2 rgbs) they are placed at 8 cms from each other with parallel optical rays. You can ignore the right or left or mix them.