NVlabs/Deep_Object_Pose #294: Questions Regarding inference file generation

ArghyaChatterjee, 2023-04-08:
Hi,

I had a question regarding the inference file generation after the training. I used the train.py script from train2 folder. I can see the file looks different than the default one. What's the reason for that  and is that a problem ? The one that I generated is ironbar_02.pth and the ones that were already there are the other ones that I downloaded, they look different than the one that is generated by me.

[image]

TontonTremblay, 2023-04-09:
I am not sure why they are different visually. It is a good question. Maybe torch changed the way it dealt with pth filing.

TontonTremblay, 2023-04-09:
if it is still running I would say do not worry about it.

ArghyaChatterjee, 2023-04-09:
Great, thanks for the confirmation.

closed 2023-04-09 by ArghyaChatterjee