NVlabs/Deep_Object_Pose #298: Making the format of the dataset

JYX0118, 2023-04-26:
Hello, are there any formatting requirements for creating my own dataset, and I look forward to your reply!

TontonTremblay, 2023-04-26:
What do you mean? There is an example json file you can look into if you want to know more. https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/output/output_example/00000.json The training script needs similar data to this.

JYX0118, 2023-04-27:
Thank you for your answer, I still have the following questions:
1. If I need to identify my own objects and I have to make my own dataset, which tool is better to use to make the dataset.
2. What should I pay attention to when training my own dataset, for example, should I use scripts/train.py or scripts/train2/train.py, and do I need to modify other places?
3、When I have trained my own dataset and got the model, which folder should I put the model into? Then launch camera.launch and dope.launch, I know there must be some parameters that need to be modified, but I'm not quite sure which places need to be modified.
Looking forward to your answer, thanks again!

TontonTremblay, 2023-04-27:
1. Nvisii right now is the most complete tool you should use, check the
script for generating training data.
2. train2 has an output in the tensorboard, you need to see belief maps
being visible.
3. Follow the instructions on how to use the config file with respect to
object sizes.

I hope this helps.