NVlabs/Deep_Object_Pose #299: (Question) - How to train the model ?

TapleFlib, 2023-05-05:
I'm new to deep learning and stuff , I saw the wiki on how to train , but I'm not sure which one to do first , anyone can give me steps to follow ? for example :
1. Download NDDS (Idk if it's an app or what?)
2. Put images
3. train using python file?
4. get weights file?
5. run the dope launch file?

I'm using ubuntu 20.04 , can I train it in windows or linux ? 
Thankyou

TontonTremblay, 2023-05-05:
What are you trying to do?

TontonTremblay, 2023-05-10:
do you have a 3d model?

TapleFlib, 2023-05-11:
currently, no. I can make 1 if it's necessary

TapleFlib, 2023-05-11:
Any detailed steps from the start until I get the weights file?

TontonTremblay, 2023-05-11:
Read the readme(s)

TapleFlib, 2023-05-12:
I read it but it doesn't include steps for training custom object.

TontonTremblay, 2023-05-12:
data:
https://github.com/NVlabs/Deep_Object_Pose/tree/master/scripts/nvisii_data_gen
training:
https://github.com/NVlabs/Deep_Object_Pose/tree/master/scripts/train2

TapleFlib, 2023-05-15:
Thankyou ! I'll try it .