NVlabs/Deep_Object_Pose #300: Regarding annotated dataset generation pipeline for dope and centerpose

ArghyaChatterjee, 2023-05-10:
Hello,

I have a question regarding dataset generation process for DOPE and Centerpose. For DOPE, it seems like the dataset is generated either by NDDS or NViSII but for CenterPose, you used objectron 3D annotated dataset. Is there a particular reason why you didn't use your own annotated synthetic data generation pipeline for CenterPose that you used for DOPE ?

TontonTremblay, 2023-05-11:
Great observation, generating a dataset of classes is not as straightforward as generating a dataset for a single object. I could have use shapenet though, but the overlap is not amazing with objectron. I have trained internally centerpose on synthetic data and the perf is similar to DOPE.

ArghyaChatterjee, 2023-05-11:
Thanks for the clarification. I got the answer. Closing the issue.

closed 2023-05-11 by ArghyaChatterjee