NVlabs/Deep_Object_Pose #302: About 3d models to gen data

huyle24022002, 2023-06-13:
Hi, i'm a newbine in this field, i want to ask about a model 3d to generate data, how can we get it is it 3d scanning, if it true, i have a problem about my object. I have use my phone to scan it and config as Cherries folder but when data generate my object to small and didn't have color can you help me solve this problem? Thank you very much

huyle24022002, 2023-06-14:
> Do you have some examples you can share with us, I think I understand that you have a 3d model you scanned with your phone, but when you put in nvisii you get a too small object. Did you try changing the scale?
Hi TonTonTremblay, in this screenshot is my 3d scan object file (right) and Cherries exam object (left), as I see that my file didn't have 3 file like example is ._texture_map, .bin file and json file. How can i generate these file. Thank you for your help

huyle24022002, 2023-06-14:
> Do you have some examples you can share with us, I think I understand that you have a 3d model you scanned with your phone, but when you put in nvisii you get a too small object. Did you try changing the scale?

Hi TonTonTremblay, in this screenshot is my 3d scan object file (right) and Cherries exam object (left), as I see that my file didn't have 3 file like example is ._texture_map, .bin file and json file. How can i generate these file. Thank you for your help
[image]

TontonTremblay, 2023-06-14:
Can you show me the 3d model using something like meshcat?

huyle24022002, 2023-06-14:
> Can you show me the 3d model using something like meshcat?
hmmm i use Kiri Engine in my phone to scan object and all these file is from the software

TontonTremblay, 2023-06-14:
https://www.meshlab.net/ install this. Double click on the .obj file or open the file with this, sorry not meshcat but meshlab.

huyle24022002, 2023-06-14:
> https://www.meshlab.net/ install this. Double click on the .obj file or open the file with this, sorry not meshcat but meshlab.

Thank you very much, i will try it tomorow, if i have some problem, may i ask you again in this post?

TontonTremblay, 2023-06-14:
for sure any question is welcome. This will only give you the visual of the object. I am sure your problem is scale.