NVlabs/Deep_Object_Pose #303: can this paper be reproduced using Gazebo

flashmoment, 2023-06-22:
I am a beginner, can this paper be reproduced using Gazebo, and have the developers also done simulations?

flashmoment, 2023-06-22:
This is the first paper in the field of machine vision grasping that I want to reproduce, and I have many basic questions：
Does your algorithm include the joint control part of the robotic arm, that is, calculating the joint parameters of the robot during grasping, so that the parameters can be passed to ROS for publishing later？

TontonTremblay, 2023-06-22:
This work does object pose estimation, you could generate your data using gazebo. 

The grasping points with this are hard coded and we used RMPs to control the robot.

flashmoment, 2023-06-23:
tks,Are you saying that the grasp points are already written in the code based on the object’s pose and classification, rather than generated by an algorithm?

flashmoment, 2023-06-23:
> This work does object pose estimation, you could generate your data using gazebo.
> 
> The grasping points with this are hard coded and we used RMPs to control the robot.

tks,Are you saying that the grasp points are already written in the code based on the object’s pose and classification, rather than generated by an algorithm?

TontonTremblay, 2023-06-23:
We did not put the code for grasping as part of this release. I could walk you through what we did though if you are interested, but I do not think I would share the code though.

flashmoment, 2023-06-24:
> We did not put the code for grasping as part of this release. I could walk you through what we did though if you are interested, but I do not think I would share the code though.我们没有将抓取的代码作为此版本的一部分。如果你感兴趣的话，我可以带你看看我们做了什么，但我不认为我会分享代码。
Thank you. I understand that there are certain things that are not convenient to disclose. I am very interested in your proposal and eagerly await your response.