NVlabs/Deep_Object_Pose #304: I have used DOPE for the pick and place task

Linhhongduc, 2023-06-24:
This is a great topic, thank you for creating it. From what you guys have done, I have created an MFC interface to simulate and control the robot. In addition, I use the zed camera to measure the depth to the center of the object, calculate the position and orientation of the object in robot coordinates. From there I was able to pick up the object and place it in the correct position on the tray placed on the conveyor.
Here is a video connecting simulation software with robot gripper and DOPE
https://github.com/NVlabs/Deep_Object_Pose/assets/124419166/237dd87f-0de2-4283-8b47-425b23bcbfc4

This is a video of a robot that picks and places 5 objects in a random position
https://github.com/NVlabs/Deep_Object_Pose/assets/124419166/c2687313-49e4-46fb-900c-950218e84421

TontonTremblay, 2023-06-26:
These are mighty impressive results and very cool to watch. What is the robot you are using? As for our experience, I defined 8 pre grasp points that are axis-aligned in the object coordinate frame. When the pose is then put in the robot frame, we get then use the grasping point that is at the highest z coordinate. We simply go there and close the gripper.

Linhhongduc, 2023-06-27:
I use the old Denso robot, because this is a university environment, the equipment is not too modern. My goal is to pick up the item and place it in the tray on the conveyor in an upright position with the can lid facing up. So I need to pick up at the center of the can and control the robot with different postures to ensure the above goal😄

TontonTremblay, 2023-06-27:
This is very cool!