NVlabs/Deep_Object_Pose #306: How to train the model on Colab

huyle24022002, 2023-06-27:
Hi, i have asked about datagen with my object few day ago, now i have successful to gen my data, but i have a problem that my local laptop is to slow to train dope model, so is there any solution to train or any tutorial to train this on free GPU as colab. Thank you very much for every support

TontonTremblay, 2023-06-27:
I am glad to hear that you can render your object. Training this model is pretty heavy, I am not aware of any platform you could use for free as I am not really aware of this, collab sounds promising, since the script is one file you should be able to hack it, but I have never done it personally, if you do, please share, I will add a link to it.

Okabe-Rintarou-0, 2023-06-27:
using kaggle is ok

huyle24022002, 2023-06-27:
> I am glad to hear that you can render your object. Training this model is pretty heavy, I am not aware of any platform you could use for free as I am not really aware of this, collab sounds [...]

maybe i will try it in this summer, thank you for your answer

huyle24022002, 2023-06-27:
Have you try it before, or have some tutorial about it, thank you

huyle24022002, 2023-06-27:
> using kaggle is ok

Have you try it before, or have some tutorial about it, thank you