NVlabs/Deep_Object_Pose #307: Regarding trainning my object

Okabe-Rintarou-0, 2023-06-27:
Hello, I am training my customized object, i have generated some datasets(use nb_objects to generate many instances in one scene):
[image]
after some time of training, i got some results but find it strange:
[image]

I am new to DOPE, so i have no idea what's going on, can somebody give me a hand please?

Okabe-Rintarou-0, 2023-06-27:
here are some images generated using --save flag
[image]
[image]

mintar, 2023-06-27:
Did you specify the symmetries? See here: https://github.com/NVlabs/Deep_Object_Pose/tree/master/scripts/nvisii_data_gen#handling-objects-with-symmetries

Also please show a visualization of the belief maps.

TontonTremblay, 2023-06-27:
To echo @mintar you need to specify the symmetries to get stability.

Okabe-Rintarou-0, 2023-06-28:
Thank you for your replies. I have specified the symmetries through model_info.json:
```json
{
  "symmetries_continuous": [{"axis": [0, 1, 0], "offset": [0, 0, 0]}],
  "align_axes": [{"object": [0, 0, 1], "camera": [0, 0, 1]}, {"object": [0, 1, 0], "camera": [0, 1, 0]}]
}
```

Here is the obj file, the y-axis is the axis of rotational symmetry:
[tube.obj](https://drive.google.com/file/d/1nLqfTXnurXfytF3CWOi9uf4LaDY2L0UA/view?usp=sharing)

Here is another example with its belief map:
[image]
[image]

Okabe-Rintarou-0, 2023-06-28:
Is it because my model lacks training?

Okabe-Rintarou-0, 2023-06-28:
I want to detect the tube cap. I am wondering whether my 3d model is real enough?
[image]

Okabe-Rintarou-0, 2023-06-28:
btw, i use --scale to enlarge the object(it's very small)