NVlabs/Deep_Object_Pose #310: Train problem with train.py in script/train2

huyle24022002, 2023-07-11:
Hi, I have installed all requirement package in requirement, i also install and upgrade cuda but i still not run train.py file, there is my error. Thank you to any support <3
[image]
[image]

TontonTremblay, 2023-07-11:
Which version of python and torch?

huyle24022002, 2023-07-12:
> Which version of python and torch 
Python 3.8 and Pytorch 2.0.1+cu117

TontonTremblay, 2023-07-12:
Can you try with pytorch 1.8? I think they changed how things worked in 2.0 and we would need to update the code to work with it, I welcome PR if you end up doing it !

huyle24022002, 2023-07-12:
> Can you try with pytorch 1.8? I think they changed how things worked in 2.0 and we would need to update the code to work with it, I welcome PR if you end up doing it !

I will try it now and response soon, thank you very much