NVlabs/Deep_Object_Pose #313: How to print the positation and orientation of the detection target in world frame?

Yukizyh, 2023-07-25:
Hi, I'm trying to calculate the accuracy of my own dataset and want to use coordinates in the world frame. The json file from inference.py output contains 'location' and 'quaternion_xyzw', but they are based on the camera frame. I tried using the 'camera_view_matrix' from generating my own dataset to calculate back to world coordinates, but the results differ from what was set. How can we obtain world coordinates? Thank you!

Yukizyh, 2023-08-01:
Thanks a lot for helping! I just want to ensure that are the 'location' and the 'quaternion_xyzw' we get from "inference.py' based on the camera coordinate system which we used to generate the data?

TontonTremblay, 2023-08-01:
Inference will be in the opencv coordinate frame. So similarly apply the same transform above.

Yukizyh, 2023-08-03:
Hi, Tonton. Really thanks for your time. 

But I still have some questions. The detected result I got from 'inference.py' only contains the intrinsics of the camera and the  'location', 'quaternion_xyzw', 'projected_cuboid' of the objects. So, if I want to use the above method to calculate the mtx_obj_world, do we need to know the mtx_camera base on the opencv coordinate? If so, how can I get that? 

Because I tried to use the mtx_camera in the world coordinate system and do the calculation with the object position and rotation that we detected. But the obj_world_position and obj_world_quat are not the same as the value I set. (Which rotated 90 degrees by the y-axis.)

I am new to this so thanks for your patience!

TontonTremblay, 2023-08-03:
https://chat.openai.com/share/a9e30c79-f70a-456d-8e54-2ded88bacab0 here is a session that I just did that produces the solution you need. 

Did you try to visualize the results inference.py produces with https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/metrics/render_json.py this code I think answers most of your questions.