NVlabs/Deep_Object_Pose #316: data for train2.py

hsisfree, 2023-09-04:
I'm using tarin2.py. I downloaded hope 3D models at https://drive.google.com/drive/folders/1jiJS9KgcYAkfb8KJPp5MRlB0P11BStft.
But I can't find a way how to run train2.py with these models.
I tried to use nvisii_data_gen but I stuck in how to use hope 3D models.
I think I have to change script but I don't know how...
Please help.

hsisfree, 2023-09-04:
> _No description provided._

TontonTremblay, 2023-09-04:
I think there are some data examples, but you would have to generate your own data.

hsisfree, 2023-09-05:
I downloaded data from https://github.com/NVlabs/Deep_Object_Pose/issues/155#issuecomment-791148200.
And I run python -m torch.distributed.launch --nproc_per_node=1 train.py --network dope --epochs 2 --batchsize 10 --outf tmp/ --data ../nvisii_data_gen/output/output_example/
command. But I got error like this
[image]

hsisfree, 2023-09-05:
> I think there are some data examples, but you would have to generate your own data.

I downloaded data from https://github.com/NVlabs/Deep_Object_Pose/issues/155#issuecomment-791148200.
And I run python -m torch.distributed.launch --nproc_per_node=1 train.py --network dope --epochs 2 --batchsize 10 --outf tmp/ --data ../nvisii_data_gen/output/output_example/
command. But I got error like this
[image]

closed 2023-10-09 by hsisfree