NVlabs/Deep_Object_Pose #319: You do not need multi gpus, the code uses the distributed package from pytorch, so you need to start it with the distributed packages starting lines. I think with a 3050 you might need to reduce quite a bit the batchsize.

hsisfree, 2023-09-12:
You do not need multi gpus, the code uses the distributed package from pytorch, so you need to start it with the distributed packages starting lines. I think with a 3050 you might need to reduce quite a bit the batchsize.

_Originally posted by @TontonTremblay in https://github.com/NVlabs/Deep_Object_Pose/issues/318#issuecomment-1714032759_

I tried to run. When I use python -m torch.distributed.launch --nproc_per_node=1 train.py --network dope --epochs 2 --batchsize 1 --outf tmp/ --data ./example command, I got RuntimeError: No rendezvous handler for env://
I think this has happened cause I'm using window 10.  Or is there anoter reason??

TontonTremblay, 2023-09-12:
Hard for me to know, it might also be that a 3050 at the end of the day is not the best equipped to do deep learning.

blaine141, 2023-09-20:
Sounds like it is maybe something with your environment. Try updating your drivers, packages, and maybe even the operating system. Maybe switch to Linux. It is what I use.

Also, check and make sure you are not running out of memory on your GPU

hsisfree, 2023-10-09:
Thanks for answer. I will check my drivers and try to train again.

closed 2023-10-09 by hsisfree