NVlabs/Deep_Object_Pose #320: i met a problem that can not be solved a long time

100656cyx, 2023-09-26:
Traceback (most recent call last):
  File "/root/miniconda3/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/root/miniconda3/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/root/miniconda3/lib/python3.8/site-packages/torch/distributed/launch.py", line 260, in <module>
    main()
  File "/root/miniconda3/lib/python3.8/site-packages/torch/distributed/launch.py", line 255, in main
    raise subprocess.CalledProcessError(returncode=process.returncode,
subprocess.CalledProcessError: Command '['/root/miniconda3/bin/python', '-u', 'train.py', '--local_rank=0']' returned non-zero exit status 1.

TontonTremblay, 2023-09-26:
With the latest version of pytorch, they changed how distributed works, we would need to update the code, you can probably use pytorch 0.4.

100656cyx, 2023-09-27:
Thank you for the reply!

So could you please send me the complete code that is available to run? 
This code may help me a lot!
I have some ideas about improving and I have already get my own datasets containing 10k pngs

Thanks sincerely!

发自我的iPhone

100656cyx, 2023-09-27:
Thank you for the reply!

So could you please send me the complete code that is available to run?
This code may help me a lot!
I have some ideas about improving and I have already get my own datasets containing 10k pngs

Thanks sincerely!

抄送: "X ***@***.******@***.***&gt;;