NVlabs/Deep_Object_Pose #325: Regarding custom data generation

ArghyaChatterjee, 2023-10-23:
Hello, 

I was trying to generate dataset for centerpose using your (dope) pipeline. There are 4 problems that I am facing.

1. I have taken some images at different exposures to merge and generate some hdr images which represents my lab's background. Now, my hdr images are 1920 x 1080 in resolution.  The images that I am generating using your pipeline is 1280 x 720 in resolution. Now when I try to generate a training dataset with the main image and the distractors, I can see the images taken and the corresponding annotated dataset were generated but the background images are zoomed in (which is not representative of the original background). I have tried to change the position of the camera and the fov, but that results into distorting the image itself. How can I change this thing ?

Normal (Without changing anything in your script, auto zoomed in background which is a problem):
[image]

Camera Eye changed (from 'eye':visii.vec3(0,0,0) to 'eye':visii.vec3(0,0,-2), looks distorted):
```# data structure
random_camera_movement = {
    'at':visii.vec3(1,0,0),
    'up':visii.vec3(0,0,1),
    'eye':visii.vec3(0,0,-2)
}
```
[image]

Camera fov changed to 2 (from default 0.78 to 2, looks distorted):
```
    camera = visii.entity.create(
    name = "camera",[00000](https://github.com/NVlabs/Deep_Object_Pose/assets/28845357/0aaef259-8281-4379-8d5b-d2fce87e9eb9)
    transform = visii.transform.create("camera"),
    camera = visii.camera.create_perspective_from_fov(
    name = "camera",
    field_of_view = 1.5,
    aspect = float(opt.width)/float(opt.height)
    )
```
[image]

2. In the original objectron dataset that centerpose is trained on, it contains keypoints 3d and scale of object in the corresponding annotated json file. As dope doesn't need that information, you haven't included that inside the nvisii interface. Can you tell me how to generate the information for centerpose dataset ?? 
Here is how the json file looks like for dope:
```
{
    "camera_data": {
        "camera_look_at": {
            "at": [
                1.0,
                0.0,
                0.0
            ],
            "eye": [
                0.0,
                0.0,
                0.0
            ],
            "up": [
                0.0,
                0.0,
                1.0
            ]
        },
        "camera_view_matrix": [
            [
                0.0,
                0.0,
                1.0,
                0.0
            ],
            [
                -1.0,
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                -1.0,
[... 733 lines cut ...]
                0.6805735602451516
            ],
            "scale": [
                0.12999999523162842,
                0.14000000059604645,
                0.09000000357627869
            ]
        }
    ]
}
```

3. When it's generating the annotated dataset, why can't I see the segmented.exr images ( I mean it's blank white). How to get that ?? I can only see a depth.exr which is a 32 bit binary image (which looked more like a segmentation image than a depth image). Here are the things that the pipeline is generating with the depth.exr as a binary image and seg.exr as a blank white image.

[image]

The depth image (.exr) looks like this:

[image]

The hdr image (.exr) looks like this:

[image]

4. Is there a way to interactively change the parameters inside nvisii so that I can adjust parameters visually, I mean where to put the objects, distractors and camera itself ??

ArghyaChatterjee, 2023-10-24:
@TontonTremblay, can you respond please ?

TontonTremblay, 2023-10-25:
I am in vacation this week, will answer next week.

TontonTremblay, 2023-10-30:
Wow this is some unknown territory. I am impressed by what you are trying to do, and I have always wanted to try something along these lines. Is this the tutorial you followed: https://blog.polyhaven.com/how-to-create-high-quality-hdri/ 
nvisii uses this model of hdr map, I am afraid I cannot help much more than that tbh. But please share your results here.

TontonTremblay, 2023-10-30:
I see there are questions in there :P 

1. See above, I do not think you created your hdr map correctly.
2. Yeah you are correct, you need the cuboid sizes, https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/utils.py#L950-L960 this is the information you need and you need to then normalize it by one axis. 
3. exr images are a format that is not bounded by values, you can store what ever you want in them, then most viewer wont know how to read the segmentation. https://chat.openai.com/share/dbfb8ef1-6504-48e6-bc2b-9972f18283f0 try something like this. 
4. https://github.com/owl-project/NVISII/blob/master/examples/17.materials_visii_interactive.py try to make the script work with this and then you could control your params live. 

Good luck