NVlabs/Deep_Object_Pose #327: Question regarding data generation pipeline

ArghyaChatterjee, 2023-11-02:
Hello,

As you know from some other issues, I have been trying to generate some ground truth data for centerpose using dope's pipeline. Here is a json data containing the annotation for cup model from objectron dataset (generated using the methods mentioned inside the original centerpose repo) :
```
{
    "AR_data": {
        "plane_center": [
            0.026276886463165283,
            0.03733876347541809,
            -0.42468586564064026
        ],
        "plane_normal": [
            -0.7663699388504028,
            0.09618763625621796,
            0.6351574063301086
        ]
    },
    "camera_data": {
        "camera_projection_matrix": [
            [
                1.6554118394851685,
                0.0,
                0.019000232219696045,
                0.0
            ],
            [
                0.0,
                2.2072157859802246,
                -0.004737734794616699,
                0.0
            ],
            [
                0.0,
                0.0,
                -0.9999997615814209,
                -0.0009999998146668077
            ],
            [
                0.0,
                0.0,
                -1.0,
                0.0
            ]
        ],
        "camera_view_matrix": [
            [
                -0.26714298129081726,
                -0.7525513172149658,
                -0.6019145250320435,
                -0.055233731865882874
            ],
            [
                -0.9542973041534424,
                0.11975152790546417,
                0.27381736040115356,
                0.18261873722076416
            ],
            [
                -0.13398146629333496,
                0.6475539207458496,
                -0.7501484751701355,
                -0.00018225116946268827
            ],
            [
                0.0,
                0.0,
                0.0,
                1.0
            ]
        ],
[... 430 lines cut ...]
                "fx": 521.6779174804688,
                "fy": 521.6779174804688,
                "cx": 640.0,
                "cy": 360.0,
                "s": 0.0
            },
            "captured_image_size": {
                "width": 1280,
                "height": 720
            }
        }
    ]
}
```

2. In the centerpose json data, Inside AR_data field, there is plane_center and plane_normal which is missing in the synthetic data generation pipeline. Also, the keypoints_3d ( actual location) value is missing in the synthetic datasets. Are these values needed if I am trying to train a single instance of mug which has no axis symmetry for centerpose ??

3.  In vast amount of the centerpose annotation json data, you have the z value of the "location" field to be negetive like this:
 `-0.38022490531119946` but for the synthetic data generation pipeline, the data is more or less bounded by a volume like this: 
``` 
xmin = -0.75, xmax = 0.75, 
ymin = -0.75, ymax = 0.75 
z min = 0.15, zmax = 2.0  
```
where z value is never negetive for synthetic data generation pipeline (which also makes sense as in front of camera, the z depth should be positive not negetive). Why is the discrepency and how to mitigate through this problem ??

TontonTremblay, 2023-11-02:
1. Nvisii has a set camera intrinsics that uses opencv style cam int. You should be able to change that. 
2. Depends what you want to do, you could export the 3d poses, use the `cuboid_` child of the object you want to export to export their 3d position, follow the same workflow as the main object. 
3. yeah the z value is negative because the camera is an opengl camera. 
[image]
You can see that everything in front of the camera will have z negative, and if you pay attention the y should also be inverted. Check chatgpt on how to go from opengl coordinate frame to opencv. 

I think overall you are on the right direction and you are making good progress, good job. advice with 3d, take your time and make one change at the time and then debug it. Easy to introduce errros that are hard to recover from.

ArghyaChatterjee, 2023-11-06:
Hello @TontonTremblay ,

Thanks for the reply. I understood 1 & 3 but couldn't understand point 2. 

I can see the https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/single_video_pybullet.py#608
in this line, you have commented out the cuboids. I don't see any definition for this one inside the single_video_pybullet.py file. 

Inside the utils.py file, in the export_to_ndds_files function definition, it's set to none: https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/utils.py#1103.

I do see some cuboid definitions inside the utils.py file as add_cuboid here: https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/utils.py#913

And here: https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/utils.py#998

Lastly, I see a logic here: https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/utils.py#1233
```
        if not cuboids is None:
            cuboid = cuboids[obj_name]
        else:
            cuboid = None
```

I am somewhat confused as there are some things parallely going on as well as I need those cuboid keypoints with respect to camera frame (not world frame). To ensure everything is on right track, it's important that I don't do anything wrong.

ArghyaChatterjee, 2023-11-06:
Also, just wanted to confirm that have  you used opengl camera convention for centerpose prediction ?? I believe centerpose is using objectron dataset and in that logic, objectron dataset is also following opengl convention. If that's the case, inside utils.py, if I commented out the cam_matrix 180 degree rotation part, and add camera_projection_matrix like the same way as cam_matrix to produce everything in opengl format, is that ok ??
```
# camera view camera
    cam_matrix = visii.entity.get(camera_name).get_transform().get_world_to_local_matrix()

    # rotate camera by 180° around x
    # from: X right, Y up and Z out of the image towards the viewer
    # to: X right, Y down and Z into the image away from the viewer (= OpenCV / NDDS / ROS "optical" frame)
    # cam_matrix = visii.mat4(1, 0, 0, 0,
    #                         0, -1, 0, 0,
    #                         0, 0, -1, 0,
    #                         0, 0, 0, 1) * cam_matrix

    cam_matrix_export = []
    for row in cam_matrix:
        cam_matrix_export.append([row[0],row[1],row[2],row[3]])
    
    # camera_projection_matrix
    cam_proj_matrix = visii.entity.get(camera_name).get_camera().get_projection()

    cam_proj_matrix_export = []
    for row in cam_proj_matrix:
        cam_proj_matrix_export.append([row[0],row[1],row[2],row[3]])
```

After that, here is how my json file was updated:
```
{
    "camera_data": {
        "camera_look_at": {
            "at": [
                1.0,
                0.0,
                0.0
            ],
            "eye": [
                0.0,
                0.0,
                0.0
            ],
            "up": [
                0.0,
                0.0,
                1.0
            ]
        },
        "camera_projection_matrix": [
            [
                0.8151217699050903,
                0.0,
                0.0,
                0.0
            ],
            [
                0.0,
                1.4491053819656372,
                0.0,
                0.0
            ],
            [
                0.013498933985829353,
                0.014036557637155056,
                -1.0010005235671997,
                -1.0
            ],
            [
                0.0,
                0.0,
                -0.10005002468824387,
                0.0
[... 132 lines cut ...]
            ],
            "provenance": "nvisii",
            "px_count_all": 0,
            "px_count_visib": 0,
            "quaternion_xyzw": [
                0.013572394847869873,
                0.15302777290344238,
                -0.21785865724086761,
                -0.9638134241104126
            ],
            "quaternion_xyzw_worldframe": [
                -0.28967729210853577,
                0.6605637073516846,
                0.456277459859848,
                -0.5211083292961121
            ],
           "scale": [
                0.12563899159431458,
                0.09907500445842743,
                0.09066499769687653
            ]
        }
    ]
}
```

TontonTremblay, 2023-11-07:
When we did objectron, I am afraid I do not remember exactly how we parsed the data. So I am afraid I wont be able to help, but normally we train algorithms on opencv conventions to make our life a little easier on the robot. 

But to answer your question, yes, if you remove the rotation you will get the camera pose in opengl coordinate frame. nvisii uses opengl coordinate frames (and going to opencv is pretty simple). 

In the script when you pass the cuboid data structure, you will export the cuboid in its local frame, I did not include it in the final version of the code to avoid confusion with the exported data with the projected_cuboid. But also it is a data structure to keep track of. If you check the add_cuboid in the utils, it returns the cuboid in the object coordinate frame, you can use that to export the normalize cuboid centerpose needs. I hope this helps. https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/nvisii_data_gen/utils.py#L996 I think the cuboid to export_ndds needs object_name:cuboid dictionary.