NVlabs/Deep_Object_Pose #328: Not displaying image in Rviz

bjoshai, 2023-11-03:
$ roslaunch dope camera.launch 
there is no error with this
$  roslaunch dope dope.launch [config:=/path/to/my_config.yaml]
got a warning as this
  warnings.warn(
/home/cone/.local/lib/python3.8/site-packages/torchvision/models/_utils.py:223: UserWarning: Arguments other than a weight enum or `None` for 'weights' are deprecated since 0.13 and may be removed in the future. The current behavior is equivalent to passing `weights=None`.
  warnings.warn(msg)
  
  and i am getting topics as in pic
[image]

and  i am ruuning rviz after adding image i am not gettig image -NO IMAGE
[image]

can anyone help me with this?

TontonTremblay, 2023-11-03:
I cannot see the image topic you are seeing, if you have a little amount of experience with ros, I would recommend to do a few tutorials with ros and rviz. If DOPE is running correctly you will be able to select the topic on the menu on the left.

bjoshai, 2023-11-04:
i am getting image but not other things
[image]
help me in this

TontonTremblay, 2023-11-04:
Which topic did you put?

bjoshai, 2023-11-04:
[image]
/rgb_points--->image

TontonTremblay, 2023-11-04:
If you put a soup can now?

bjoshai, 2023-11-05:
[image]
added the  images rgb_points and belief_soup image is they are visible after that i am not getting anything for pose and marker array .

TontonTremblay, 2023-11-05:
everything seems to be working well :P but an object in front now that is the soup.

bjoshai, 2023-11-06:
> everything seems to be working well :P but an object in front now that is the soup.

i didnt get ,could you please elaborate it?
Thanks

bjoshai, 2023-11-06:
i didnt get ,could you please elaborate it?
Thanks

closed 2023-11-06 by bjoshai

TontonTremblay, 2023-11-07:
You need the physical object of a soup, google search an image of the tomato soup and point the camera at your screen.