NVlabs/Deep_Object_Pose #329: Error while running single_video_pybullet.py from nvisii_data_gen

umegan, 2023-11-03:
I'm trying to create own dataset for issac ros dope using nvisii_data_gen.
I running on [DOPE in a Docker Container](https://github.com/NVlabs/Deep_Object_Pose/tree/master/docker#dope-in-a-docker-container) and have installed all the requirement.txt for nvisii_data_gen.

**Script**
`python3 single_video_pybullet.py --nb_frames 1 --scale 0.01`

**Error**
```
root@anpc:~/catkin_ws/src/dope/scripts/nvisii_data_gen# python3 single_video_pybullet.py --nb_frames 1 --scale 0.01
pybullet build time: May 20 2022 19:44:17
folder output/output_example// exists
[ 2][    DENOISER]: Unable to load denoiser weights
Optix call (optixDenoiserSetModel(OptixData.denoiser, kind, nullptr, 0)) failed with code 7990 (line 1887)
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
Aborted (core dumped)
root@anpc:~/catkin_ws/src/dope/scripts/nvisii_data_gen# 
```

I would appreciate any guidance on resolving this error or suggestions on what I may have missed during the setup process.

Thank you

TontonTremblay, 2023-11-03:
Humm that is a new one. I never seen this error, can you tell me more about your docker? I would recommend to use this docker: https://github.com/owl-project/NVISII#docker instead. Nvisii and docker is not easy to get to work, also your work machine NVIDIA drivers will have an impact on it.

umegan, 2023-11-10:
Thanks for the respone!
This is some information about my docker and system:
OS: Ubuntu 22.04.3 LTS
GPU: NVIDIA GeForce RTX 4080
Driver version: 535.113.01
CUDA version: V11.5.119
Docker version: 24.0.7

About your recommend docker ( https://github.com/owl-project/NVISII#docker ) , It seem like the base docker image use to built dockerfile not exist anymore

`FROM nvidia/cuda:10.2-runtime-ubuntu18.04`

TontonTremblay, 2023-11-10:
ahhh that sucks. Yeah nvisii does not work on 2204 atm. Maybe we will get to update the code base. Did you try to pass the missing .so files in the docker container? You need an image that is 1804 that is for sure, the rest you can use an other. Recently I have two machine one with 1804 and one 2204. But that might not help you out. I am sorry I think you are going to need to fiddle around.  
```
[ 2][    DENOISER]: Unable to load denoiser weights
```
it looks like your are missing something from the drivers.

umegan, 2023-11-20:
Yeah, look like 2204 was the problem. After change to 1804 machine everything work well. 
closing!

closed 2023-11-20 by umegan