NVlabs/Deep_Object_Pose #330: Adding a background to single_video_pybullet.py

rpapallas, 2023-11-03:
Hello,

Thank you for sharing this great work and for maintaining the code. I am interested in generating my own data to train DOPE with our own models. I find the scripts for doing this (i.e., nvisii_data_gen) very useful.

I would like to generate data with a certain background. The background is a bin with many other objects and I would like to train DOPE to be able to correctly identify objects of interest among cluttered environments, like a bin. I took a picture (in JPG format) of the bin at a high resolution.

However, I can't figure out how to put the background image properly to the scene. I try to do via the "dome" and by renaming my jpg background image to .hdr under "dome_hdri_haven". The script picks up the image just fine and loads it as a background but it's zoomed in and I can't seem to find a proper way to zoom out.

Any idea how this bin background could be achieved easily with your script?

Thank again.

TontonTremblay, 2023-11-03:
what I would do is put the image on a large square (rectangle plane). Something like this: https://github.com/owl-project/NVISII/blob/master/examples/10.light_texture.py#L42-L56, you could use 2 one behind the camera as well to provide light source. I am not a 100%. But I would try to test this out with just normal hdri haven, I would be surprise you need your background.

rpapallas, 2023-11-13:
Thank you for your help, Jonathan.

I ended up creating a 3D object with texture and included that object in the scene. Thank you.

closed 2023-11-13 by rpapallas