NVlabs/Deep_Object_Pose #332: cv2.solvepnp failed with an error

dahuahy, 2023-11-11:
The Hikang camera I use cannot be calibrated internally, is it because of this reason? The environment built is the same as that of the author.thank you

TontonTremblay, 2023-11-11:
You can use a checkerboard to get the camera calibration, would that work?

dahuahy, 2023-11-13:
After completing camera calibration, this error occurs when turning on the camera

Running DOPE... (Listening to camera topic: '/usb_cam/image_raw')
Ctrl-C to stop
[ERROR] [1699867100.066902]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7fca43889220>>
Traceback (most recent call last):
File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 750, in _invoke_callback
cb(msg)
File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 76, in callback
self.signalMessage(msg)
File "/opt/ros/noetic/lib/python3/dist-packages/message_filters/__init__.py", line 58, in signalMessage
cb(*(msg + args))

TontonTremblay, 2023-11-13:
I am not sure what this error is!  I am sorry I do not think I can help you.

dahuahy, 2023-11-14:
I solved yesterday's problem, but the following situation occurred: the terminal output is like this, but the image does not have a recognition box。

Ctrl-C to stop
4 valid points found
7 valid points found
9 valid points found
9 valid points found
6 valid points found
7 valid points found
4 valid points found
4 valid points found
6 valid points found
6 valid points found
7 valid points found
6 valid points found

TontonTremblay, 2023-11-14:
which topic are you looking at? Did you check the belief maps?

dahuahy, 2023-11-17:
I don't know what that means of belief maps,I'm a novice. When I start the terminal, I put in roslaunch dope dope.launch and I put out 4 valid points found 7 valid points found  9 valid points found...  Where in which file should I change。thank you very much

TontonTremblay, 2023-11-17:
Check rviz to visualize different images, dope outputs different views, it is a little out of the scope of this repo to teach ros.