NVlabs/Deep_Object_Pose #335: There is a significant difference in accuracy.

chenin-wang, 2023-12-04:
Despite the bounding boxes (bbox) being visually close in the image, there is a significant difference in their locations.How can this issue be improved?

[image]

[image]

chenin-wang, 2023-12-04:
Additionally, I have encountered another issue where the camera can only detect objects within a very narrow distance range. Once the distance becomes too small or too large, detection fails. I have already adjusted the 'location' field values in my dataset across a wide range. I would appreciate any suggestions you could provide. Thank you very much!

TontonTremblay, 2023-12-05:
can you put 60,60,60. I am not sure what the scale you used in your dataset. But play with this, you should be able to get to the right location. Also the quaternion might not look the same, isaacsim is probably in opengl coordinate frame, DOPE uses opencv.

chenin-wang, 2023-12-11:
Thank you very much for your response. I will conduct the testing.

TontonTremblay, 2023-12-11:
I will share this thread with the person that did the isaac-sim port.

chenin-wang, 2023-12-11:
Let me clarify my question again.I am facing an issue with object detection in a real-world scenario when adjusting the distance between the camera and the target. Despite resizing the dataset images to 1080x1080 pixels and configuring MIN_DISTANCE as 1.0, MAX_DISTANCE as 1.2, and OBJECT_SCALE as 0.002, the detection performance does not meet expectations. Previously, you suggested transforming it to 60cmx60cmx60cm. Could you please clarify if you meant resizing the physical dimensions of the object to 60cm or adjusting the OBJECT_SCALE value within the dataset so that the object appears bigger in the image? Additionally, the actual size of my object is only 6cmx6cmx6cm.
this is my Isaac config file for dope， please check it.
[dope_config.md](https://github.com/NVlabs/Deep_Object_Pose/files/13631627/dope_config.md)

andrewyguo, 2024-01-16:
@chenin-wang, could you elaborate more on why `OBJECT_SCALE` is `0.0035`? 

What happens if you change this part of the config to: 
```
OBJECT_SCALE:
- 1
- 1
- 1
OBJECT_MASS: 1
```