NVlabs/Deep_Object_Pose #336: how can I get location and quaternion of object in wordframe?

hsisfree, 2023-12-04:
Hi. I got a result using DOPE. But I think location and quaternion value is camera location and orientation in world frame. I want to get objects location and quaternion in worldframe. How can I get it?
[image]

mintar, 2023-12-05:
The easiest way is to use ROS and [TF](http://wiki.ros.org/tf/Tutorials). It's a bit of a learning curve, but absolutely worth it for solving problems such as yours. Since this is more of a TF question than a DOPE question, the better place to ask is https://robotics.stackexchange.com/ .

TontonTremblay, 2023-12-05:
You could also use mescat or open3d to debug your poses. Dont forget the different coordinate frames you are dealing with.