NVlabs/Deep_Object_Pose #337: No output in out_experiment folder

Liliva13, 2023-12-15:
I try to run inference.py without any camera and Ros. 
The model can be successfully loaded but there's no result in the 'out experiment' folder.
Could you guys suggest me some possible reasons......may cause such problem?

The line I put in terminal:
' python3 inference.py --data /input/111.png '

TontonTremblay, 2023-12-15:
If I remember correctly --data is a folder and not an image. Remove `111.png` from your query.

Liliva13, 2023-12-18:
Very sorry for the late reply @TontonTremblay 
I change it to the folder name but still no output there. 
Can the "inference.py" be run independently without being trained? Or should I train my own custom data first?

TontonTremblay, 2023-12-18:
which data are you trying, which weights, this is a very old approach to pose estimation. This needs weights for a specific object.

Liliva13, 2023-12-18:
Thank you very much！！！ I just tried again and got an output.
but the inference results are not that good, I am keep looking into it 😄
my weight used the cracker.pth you provided [there](https://drive.google.com/drive/folders/1DfoA3m_Bm0fW8tOWXGVxi4ETlLEAgmcg).
I didn't download the YCB or HOPE models but took some photos myself and save them in my datafile.
@TontonTremblay

Liliva13, 2023-12-18:
Thank you very much！！！ I just tried again and got an output.
but the inference results are not that good, I am keep looking into it 😄
my weight used the cracker.pth you provided [there](https://drive.google.com/drive/folders/1DfoA3m_Bm0fW8tOWXGVxi4ETlLEAgmcg).
I didn't download the YCB or HOPE models but took some photos myself.
@TontonTremblay

TontonTremblay, 2023-12-19:
If there the promotion prints on the box are different, then it will be harder for the network to get good results.

chloe3zhc, 2025-03-25:
@TontonTremblay I try to run interence.py with "python inference.py --weights ../weights/best.pth --data ../sample_data --object cracker".
I successfully used the train.py file to train the four images in the sample_data folder and got the best.pth file, but when I used the best.pth as the weight file for interence.py, the output file was empty, but the model and image loading were normal. 
Could you tell me what might be the problem?

TontonTremblay, 2025-03-26:
@chloe3zhc you are going to need a lot more images than the 4 samples, something like 20k I would think is needed and domain randomization, I will refer you to the original paper for the details.