NVlabs/Deep_Object_Pose #338: (Question) About weight &  fine tuning

umegan, 2024-01-09:
Hi, after a week of generating data and training the model, I obtained a pretty good result on 6D pose estimation. However, it takes quite a long time, so I want to test fine-tuning on DOPE and have a question about the weights and fine-tuning.

・Do you have a list that maps object image paths to [the weight](https://drive.google.com/drive/folders/1DfoA3m_Bm0fW8tOWXGVxi4ETlLEAgmcg) 

・Does fine-tuning on objects with similar shapes yield good results?

TontonTremblay, 2024-01-09:
This is amazing results. I am so sorry for the amount of time / work you had to put into this. I am afraid that I never spent anytime trying to build a model that could easily be fine-tuned.

umegan, 2024-01-12:
Oh, I see. What about transfer learning on the pretrained DOPE model? 

> 7. Download [the weights](https://drive.google.com/open?id=1DfoA3m_Bm0fW8tOWXGVxi4ETlLEAgmcg) and save them to the weights folder, i.e., ~/catkin_ws/src/dope/weights/.

Does it learn faster on similar shapes compared to training from scratch?

TontonTremblay, 2024-01-12:
I am not sure, I never tried it, let me know if you end up trying.