NVlabs/Deep_Object_Pose #339: ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (9,) + inhomogeneous part.

thohemp, 2024-01-17:
Hello,

I am working with a custom dataset. Trainings works fine, but when I try the inference script is throws me following error:

```sh
ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (9,) + inhomogeneous part.
```
at line https://github.com/NVlabs/Deep_Object_Pose/blob/master/scripts/train2/inference/detector.py#L485.

Anyone faced the same issue?

Belief map looks good though.

[image]

[image]

TontonTremblay, 2024-01-17:
it looks like it is due to some new numpy, can you do a clean reshape, something like `.reshape(9,2)`?

TontonTremblay, 2024-01-18:
Your results look very good btw.

thohemp, 2024-01-18:
> it looks like it is due to some new numpy, can you do a clean reshape, something like `.reshape(9,2)`?

I think the problem is that in case where the error is raised, points has Nones, e.g.:

```sh
[None, None, None, None, None, (57.18119389528066, 160.5283507309974), (61.34432166522373, 181.1726155590992), (214.68325248224997, 185.83047341657362), (122.63500075032445, 205.31441894184206)]
```

TontonTremblay, 2024-01-18:
which version of opencv and numpy are you running?

thohemp, 2024-01-19:
> which version of opencv and numpy are you running?

opencv-python 4.9.0.80
numpy 1.24.4

thohemp, 2024-01-19:
Changing the numpy version to 1.17 made it work! Thanks

TontonTremblay, 2024-01-19:
amazing :D

closed 2024-03-01 by thohemp