NVlabs/Deep_Object_Pose #342: Results bounding box shifted for custom model

thohemp, 2024-03-01:
Hello,

when I use models provided with the repo (MobileNet) I receive:

[image]
[image]

But when I use a custom model (Resnet-based), I receive:

[image]
[image]

The belief maps look very similar, but the cuboid is shifted to the right-bottom. Any idea what's happening here?

TontonTremblay, 2024-03-04:
A few things could be going on here. Can you check the raw 2d points locations -- check the array values? Are they different? Are you using different intrinsics? or cuboid size? 

But I am surprised by the shift as well, it should not be this big. Are you using train2 vs train?

TontonTremblay, 2024-03-06:
1. yeah there was a bug in the first version that I manually added this upscalling thing, I dont remember everything there. 
2. humm how come it is 52, I think these 2 pixels are shifting the results possibly. The upscale it is px*8 and then pnp is ran on that. 

Did you check the raw 2d keypoints position one vs the other, without running pnp, so we can start isolating the problem.

thohemp, 2024-03-19:
The shifting was cause due to a wrong scale factor, as I changed the heat map size. 
Thank you for your time.