NVlabs/Deep_Object_Pose #343: Some problem while creating my catkin workspace and generating data

Gaben21, 2024-03-05:
First problem:
          When running the script in the nvisii_data_gen: 
                      ` python single_video_pybullet.py --nb_frames 1 --scale 0.01 --nb_distractors 0 `
          I encountered a warning:

```
          lsf@lsf:~/lsf_ws/src/dope/scripts/nvisii_data_gen$ python single_video_pybullet.py --nb_frames 1 --scale 0.01 -- 
          nb_distractors 0
          pybullet build time: Nov 28 2023 23:51:11
          folder output/output_example// exists
          00000/00001
          /home/lsf/lsf_ws/src/dope/scripts/nvisii_data_gen/utils.py:1126: RuntimeWarning: invalid value encountered in cast
          visible_object_ids = np.unique(segmentation_mask.astype(int))
```

Second problem:
           When installing ROS and creating my catkin workspace, I run the code(my OS is ubuntu 18.04LTS):
          `rosdep install --from-paths src -i --rosdistro melodic`
          then it has a problem
```
          lsf@lsf:~/lsf_ws$ rosdep install --from-paths src -i --rosdistro melodic
          ERROR: the following rosdeps failed to install
          pip: pip is not installed
```
I would appreciate any guidance on resolving this error or suggestions on what I may have missed during the setup process.

Thank you

closed 2024-05-29 by Gaben21

Gaben21, 2024-05-30:
> Did you figure it out? I think the first error / warning is about not seeing an object in the render you did.

yes，i figured it out

Tianyi20, 2024-10-02:
> First problem: When running the script in the nvisii_data_gen: `python single_video_pybullet.py --nb_frames 1 --scale 0.01 --nb_distractors 0` I encountered a warning:
>
> ``` [...]

HI, i also face the first problem. Do you know how to fix it?

`tianyi@mei-MS-7B48:~/Deep_Object_Pose/data_generation/nvisii_data_gen$ python single_video_pybullet.py --nb_frames 1 --scale 0.01 --path_single_obj /home/tianyi/Deep_Object_Pose/data_generation/models/Ketchup/google_16k/textured.obj

pybullet build time: Nov 28 2023 23:51:11
folder output/output_example// exists
Warning, create_perspective_from_fov is deprecated and will be removed in a subsequent release. Please switch to create_from_fov.
00000/00001
/home/tianyi/Deep_Object_Pose/data_generation/nvisii_data_gen/utils.py:1126: RuntimeWarning: invalid value encountered in cast
  visible_object_ids = np.unique(segmentation_mask.astype(int))
`

TontonTremblay, 2024-10-02:
what is the error? I think the segmentation_mask is not rendered correctly.