NVlabs/Deep_Object_Pose #345: Problem about

Gaben21, 2024-03-06:


TontonTremblay, 2024-03-06:
wow transparent stuff is hard! I would suggest to use blender directly to generate your data. We are currently working on a version with omniverse, you might be able to use this, but I dont have first hand experience. But nvisii has limited capacity for transparent objects. And modelling these objects you need to be precise.

Gaben21, 2024-03-08:
I created some 3D models of the transparent objects using blender and export  .obj file and .mtl file. But when I want to import the model in NVISII, I found the material applied in the object is missing. I think if the function used to load model `nvisii.import_scence` only loads .obj file and doesn't load .mtl file. It seems I can only change the parameter alpha and transmission in the code to make the object transparent.If it has another way? 
And I have another question is that where can I define the material before loading .obj file in the code. I print the material name of the object , it show 'basicMaterial'
Thank you  very much !!!

TontonTremblay, 2024-03-08:
obj mtl models do not have the capacity to express transparency. https://github.com/owl-project/NVISII/blob/master/examples/17.materials_visii_interactive.py check this and play with transmission. You need to apply that to your model, but transparency is not well supported by nvisii.

Gaben21, 2024-03-09:
OK! Thank you!

Gaben21, 2024-03-22:
@TontonTremblay 
Hi, Tonton.I have a question about how can i adjust the distance between camera and the object when generating images. Some transparent objects in the image generated seem too small to be detected, as following image shown. And if you have some other advice to improve the result, I would be extrmely grateful.
[image]
[image]

TontonTremblay, 2024-03-22:
I would change obj_scale

Gaben21, 2024-03-22:
ok，thank you very much！