NVlabs/Deep_Object_Pose #359: Weird output

joansaurina, 2024-05-27:
Hello.
I am running inference with mustard_60.pth and the detectes cube gets this strange form. Do you know why is that? Could it be fore camera parameters?

[image]
[image]

Thanks,
Joan

TontonTremblay, 2024-05-28:
I think it is using the new model of cuboid3d, @nv-jeff made some changes to the repo, I had a "if" statement in the new inference code to use the old cuboid3d description. Maybe the update removed the if. It has been a while, https://github.com/NVlabs/Deep_Object_Pose/blob/c9f5aafd3f9f329ffc1ee922d2aa00f49d0a5c07/common/cuboid.py#L81 maybe this is the condition that is causing issues.

nv-jeff, 2024-05-28:
I introduced a bug into `common/utils.py` during refactoring; the cuboid was drawn using the wrong vertex ordering. This has been fixed in checkin https://github.com/NVlabs/Deep_Object_Pose/commit/2e021c5f05727cbd5da47330a9c5ca37ef74fee5

closed 2024-05-28 by nv-jeff